bitbank2/Thermal_Printer #5: After printing the esp32 dies

felipemouradev, 2021-02-12:
After printing the esp32 dies

13:09:10.116 -> [V][BLEScan.cpp:195] start(): >> start(duration=5)
13:09:10.116 -> [D][FreeRTOS.cpp:189] take(): Semaphore taking: name: ScanEnd (0x3ffd3484), owner: <N/A> for start
13:09:10.116 -> [D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: ScanEnd (0x3ffd3484), owner: start
13:09:10.116 -> [V][BLEScan.cpp:227] start(): << start()
13:09:10.154 -> [V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 2 0x02
13:09:10.154 -> [V][FreeRTOS.cpp:63] wait(): >> wait: Semaphore waiting: name: ScanEnd (0x3ffd3484), owner: start for start
13:09:10.154 -> [V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
13:09:10.154 -> [V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
13:09:10.154 -> [V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 7 0x07
13:09:10.190 -> [V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
13:09:10.190 -> [V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
13:09:10.190 -> [V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
13:09:10.190 -> [V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
13:09:10.190 -> [V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
13:09:10.226 -> [D][BLEAdvertisedDevice.cpp:418] setRSSI(): - setRSSI(): rssi: -79
13:09:10.226 -> [V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0xff
13:09:10.226 -> [D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0xff (), length: 25, data: 4c00021550765cb7d9ea4e2199a4fa879613a4924ce0f4e6ce
13:09:10.226 -> [D][BLEAdvertisedDevice.cpp:395] setManufacturerData(): - manufacturer data: 4c00021550765cb7d9ea4e2199a4fa879613a4924ce0f4e6ce
13:09:10.264 -> [V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
13:09:10.264 -> [V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
13:09:10.264 -> [V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
13:09:10.264 -> [D][BLEScan.cpp:97] handleGAPEvent(): Ignoring 90:32:4b:18:77:34, already seen it.
13:09:10.264 -> [V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
13:09:10.264 -> [V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
13:09:10.301 -> [V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
13:09:10.301 -> [D][BLEAdvertisedDevice.cpp:418] setRSSI(): - setRSSI(): rssi: -85
13:09:10.301 -> [V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0xff
13:09:10.301 -> [D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0xff (), length: 29, data: 060001092002101250b350a6018513aee4d47205b2e0690f17876a2ab9
13:09:10.338 -> [D][BLEAdvertisedDevice.cpp:395] setManufacturerData(): - manufacturer data: 060001092002101250b350a6018513aee4d47205b2e0690f17876a2ab9
13:09:10.338 -> [V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
13:09:10.338 -> [V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
13:09:10.338 -> [V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
13:09:10.338 -> [D][BLEAdvertisedDevice.cpp:418] setRSSI(): - setRSSI(): rssi: -43
13:09:10.374 -> [V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x1
13:09:10.374 -> [D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x01 (), length: 1, data: 1a
13:09:10.374 -> [V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0xa
13:09:10.374 -> [D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x0a (), length: 1, data: 18
13:09:10.374 -> [D][BLEAdvertisedDevice.cpp:478] setTXPower(): - txPower: 24
13:09:10.374 -> [V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0xff
13:09:10.409 -> [D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0xff (), length: 9, data: 4c0010050a18784271
13:09:10.409 -> [D][BLEAdvertisedDevice.cpp:395] setManufacturerData(): - manufacturer data: 4c0010050a18784271
13:09:10.409 -> [V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
13:09:10.409 -> [V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
13:09:10.445 -> [V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
13:09:10.445 -> [D][BLEAdvertisedDevice.cpp:418] setRSSI(): - setRSSI(): rssi: -46
13:09:10.445 -> [V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x1
13:09:10.445 -> [D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x01 (), length: 1, data: 1a
13:09:10.445 -> [V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0xa
13:09:10.445 -> [D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x0a (), length: 1, data: 08
13:09:10.479 -> [D][BLEAdvertisedDevice.cpp:478] setTXPower(): - txPower: 8
13:09:10.479 -> [V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0xff
13:09:10.479 -> [D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0xff (), length: 11, data: 4c001007691fccdc70c838
13:09:10.479 -> [D][BLEAdvertisedDevice.cpp:395] setManufacturerData(): - manufacturer data: 4c001007691fccdc70c838
13:09:10.513 -> [V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
13:09:10.513 -> [V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
13:09:10.513 -> [V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
13:09:10.513 -> [D][BLEAdvertisedDevice.cpp:418] setRSSI(): - setRSSI(): rssi: -59
13:09:10.513 -> [V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x1
13:09:10.513 -> [D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x01 (), length: 1, data: 06
13:09:10.551 -> [V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0xff
13:09:10.551 -> [D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0xff (), length: 10, data: 4c0010064e1d3a3cd938
13:09:10.551 -> [D][BLEAdvertisedDevice.cpp:395] setManufacturerData(): - manufacturer data: 4c0010064e1d3a3cd938
13:09:10.551 -> [V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
13:09:10.586 -> [V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
13:09:10.586 -> [V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
13:09:10.586 -> [D][BLEAdvertisedDevice.cpp:418] setRSSI(): - setRSSI(): rssi: -40
13:09:10.586 -> [V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x1
13:09:10.586 -> [D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x01 (), length: 1, data: 0a
[... 1171 lines cut ...]
13:09:19.965 -> [V][FreeRTOS.cpp:143] give(): Semaphore giving: name: WriteCharEvt (0x3ffd5ca4), owner: writeValue
13:09:20.000 -> [V][FreeRTOS.cpp:77] wait(): << wait: Semaphore released: name: WriteCharEvt (0x3ffd5ca4), owner: <N/A>
13:09:20.000 -> [V][BLERemoteCharacteristic.cpp:576] writeValue(): << writeValue
13:09:20.000 -> [V][BLEClient.cpp:139] disconnect(): >> disconnect()
13:09:20.000 -> [V][BLEClient.cpp:145] disconnect(): << disconnect()
13:09:20.000 -> [V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 5
13:09:20.036 -> [D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
13:09:20.036 -> [V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 5
13:09:20.036 -> [V][BLEUtils.cpp:1283] dumpGattClientEvent(): GATT Event: Unknown
13:09:20.036 -> [V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 5
13:09:20.036 -> [D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
13:09:20.073 -> [V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 41
13:09:20.073 -> [D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
13:09:20.073 -> [V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 41
13:09:20.073 -> [V][BLEUtils.cpp:1283] dumpGattClientEvent(): GATT Event: Unknown
13:09:20.106 -> [V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 41
13:09:20.106 -> [D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[... cut ...]
13:09:20.143 -> [V][FreeRTOS.cpp:143] give(): Semaphore giving: name: SearchCmplEvt (0x3ffd42a4), owner: <N/A>
13:09:20.143 -> [I][BLEDevice.cpp:604] removePeerDevice(): remove: 0, GATT role client
13:09:20.143 -> [V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 1
13:09:20.143 -> [D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
13:09:20.143 -> [V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 1
13:09:20.181 -> [V][BLEUtils.cpp:1283] dumpGattClientEvent(): GATT Event: Unknown

bitbank2, 2021-02-12:
Sending me this huge dump doesn't help diagnose this issue because you haven't showed me what your code is doing. Share the sketch.

felipemouradev, 2021-02-12:
`// Select your modem:
#define TINY_GSM_MODEM_SIM800 // Modem is SIM800L

// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define SerialMon Serial
// Set serial for AT commands
#define SerialAT Serial1

// Define the serial console for debug prints, if needed
#define TINY_GSM_DEBUG SerialMon

// set GSM PIN, if any
#define GSM_PIN ""

// Your GPRS credentials, if any
const char apn[] = "timbrasil.br"; // APN (example: internet.vodafone.pt) use https://wiki.apnchanger.org
const char gprsUser[] = "tim";
const char gprsPass[] = "tim";

// SIM card PIN (leave empty, if not defined)
const char simPIN[]   = ""; 
String imei = "";

// MQTT details
const char* broker = "4.tcp.ngrok.io";                    // Public IP address or domain name
const char* mqttUsername = "";  // MQTT username
const char* mqttPassword = "";  // MQTT password

const char* topicOutput1 = "esp/commander";
const char* topicOutput2 = "esp/health";

// Define the serial console for debug prints, if needed
//#define DUMP_AT_COMMANDS

#include <Wire.h>
#include <TinyGsmClient.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
#include <Fonts/FreeSerif9pt7b.h>
#include <ArduinoJson.h>
#include <Thermal_Printer.h>
#include <time.h>

#define SCREEN_WIDTH 128 // OLED display width, in pixels
#define SCREEN_HEIGHT 64 // OLED display height, in pixels

// Declaration for an SSD1306 display connected to I2C (SDA, SCL pins)
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, -1);

#ifdef DUMP_AT_COMMANDS
  #include <StreamDebugger.h>
  StreamDebugger debugger(SerialAT, SerialMon);
  TinyGsm modem(debugger);
#else
  TinyGsm modem(SerialAT);
#endif

#include <PubSubClient.h>

TinyGsmClient client(modem);
PubSubClient mqtt(client);

// TTGO T-Call pins
#define MODEM_RST            5
#define MODEM_PWKEY          4
#define MODEM_POWER_ON       23
#define MODEM_TX             27
#define MODEM_RX             26
#define I2C_SDA              21
#define I2C_SCL              22
[... 162 lines cut ...]
    uint32_t t = millis();
    if (t - lastReconnectAttempt > 10000L) {
      lastReconnectAttempt = t;
      if (mqttConnect()) {
        lastReconnectAttempt = 0;
      }
    }
    delay(100);
    return;
  }

  long now = millis();
  if (now - lastMsg > 30000) {
    lastMsg = now;
    time_t now = time(NULL);
    int imeiLen = imei.length() + 1;
    char imeiChar[imeiLen];
    imei.toCharArray(imeiChar, imeiLen);
    char message[200];
    sprintf(message, "{\"deviceId\":\"%s\",\"current_time\":\"%ld\"}", imeiChar, now);
    mqtt.publish(topicOutput2, message);
  }

  mqtt.loop();
}`

felipemouradev, 2021-02-12:
This code works fine, but after send text to printer the esp32 dies

bitbank2, 2021-02-12:
"Dies" doesn't describe what is happening. The code above will hang after printing because it has an infinite while loop after the printing code. It also can reset the ESP32 depending on the WiFi activity (see ESP.restart). You may be running out of stack or heap space. Bluetooth (with the printer) takes a lot of RAM and WiFi (your other code) also takes a lot of RAM. Separate the parts and see if they work reliably separately. If so, then you have a memory problem.

closed 2021-02-12 by bitbank2