bitbank2/Thermal_Printer #6: PeriPage support

Gianpy86, 2021-04-27:
Hello,

I found a way to print on PeriPage printer too.

basically, for text is enough to send a hex string before sending the text you want to print!!

bitbank, is this project something you wanna keep alive? if yes, can we collaborate?

regards

Gianpy86, 2021-04-27:
```
      String startStr = hexToAscii("10fffe01");
      tpPrint((char *)&startStr);
      tpSetFont(0, 0, 0, 0, 0);
      tpPrint((char *)"12x24 plain text\n");
      ```

this works on PeriPage printer

bitbank2, 2021-04-27:
Thanks for sharing this. The high resolution version (304 dpi) looks interesting to me, but I don't have any use for it and spending $50 just to experiment with it is outside my gadget budget. Maybe add your new info to the README and I'll merge it in.

closed 2021-04-27 by bitbank2