bitbank2/Thermal_Printer #9: [issue]  ESP32 with「PAPERANG」BLE Thermal Printer

BlynkGO, 2021-08-19:
I use 「PAPERANG」BLE Thermal Printer.
And use ESP32 to connect the printer.

[image]

ESP32 can connect to the printer.
However it can't print anything .
How to solve it?

[Result when tested by Thermal_Printer's example]
[image]

Thank you.

bitbank2, 2021-08-19:
Try commenting out the graphics printing part in the example sketch and see if it's able to print the text. I'm also finding that a recent change to ESP32's BLE support has broken the "writeWithoutResponse" logic. I'm working on a fix for that. I also ordered the little "cat" printer to do some testing on alternate gfx command sets. However, I won't receive it for about a month.

BlynkGO, 2021-08-19:
When I have tried to print only text to printer, it still can't print anything.

Now I use Arduino ESP32 core 1.0.6.
Which ESP32 core is better to compile ?

Thank you.

bitbank2, 2021-08-19:
Text printing should work on any ESP32 core. I'm on Alpha-2.0.0.

BlynkGO, 2021-08-19:
I changed to ESP32 core v2.0.0 rc1.
And I used the following code to compile

```c
//
// Thermal Printer Demo
// written by Larry Bank
// Copyright (c) 2020 BitBank Software, Inc.
//
// This program is free software: you can redistribute it and/or modify
// it under the terms of the GNU General Public License as published by
// the Free Software Foundation, either version 3 of the License, or
// (at your option) any later version.
//
// This program is distributed in the hope that it will be useful,
// but WITHOUT ANY WARRANTY; without even the implied warranty of
// MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
// GNU General Public License for more details.
//
// You should have received a copy of the GNU General Public License
// along with this program.  If not, see <http://www.gnu.org/licenses/>.
//

#include <Thermal_Printer.h>
static uint8_t ucBuf[48 * 384];
#define WIDTH 384
#define HEIGHT 240

  
void setup() {
  int i;
  // put your setup code here, to run once:
  Serial.begin(115200);
  while (!Serial);
  Serial.println((char *)"Preparing image buffer...");
  tpSetBackBuffer(ucBuf, WIDTH, HEIGHT);
  tpFill(0);

  tpDrawText(0,0,(char *)"BitBank Thermal Printer", FONT_LARGE, 0);

  Serial.println((char *)"Scanning for BLE printer");
  if (tpScan("Paperang",5))
  {
    Serial.println((char *)"Found a printer!, connecting...");
    if (tpConnect())
    {
      Serial.println((char *)"Testing plain text printing");
      tpSetFont(0, 0, 0, 0, 0);
      tpPrint((char *)"12x24 plain text\r");
      tpSetFont(1, 0, 0, 0, 0);
      tpPrint((char *)"9x17 plain text\r");
      tpSetFont(1, 1, 0, 0, 0);
      tpPrint((char *)"Underlined\r");
      tpSetFont(1, 0, 1, 0, 0);
      tpPrint((char *)"Double Wide\r");
      tpSetFont(1, 0, 0, 1, 0);
      tpPrint((char *)"Double Tall\r");
      tpSetFont(1, 0, 1, 1, 0);
      tpPrint((char *)"Double Tall+Wide\r");
      tpSetFont(1, 0, 1, 1, 1);
      tpPrint((char *)"Double Tall+Wide + emphasized\r");
      Serial.println((char *)"Disconnecting");
      tpDisconnect();
      Serial.println((char *)"Done!");
      while (1) {};      
    }
  }
  else
  {
    Serial.println((char *)"Didn't find a printer :( ");
  }
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

The result on serial-monitor as the following
```
⸮Preparing image buffer...
Scanning for BLE printer
Scan Result: Name: , Address: 08:22:82:b4:4f:bb, manufacturer data: 0600010f210aa009f73ae900414d442d524144454f4e 
Scan Result: Name: Paperang, Address: fc:58:fa:33:d7:4c, manufacturer data: 0000fc58fa33d74c, serviceUUID: 0000fee7-0000-1000-8000-00805f9b34fb, txPower: 4 
A match!
fc:58:fa:33:d7:4c
Paperang
Scan Result: Name: LYWSD03MMC, Address: a4:c1:38:8c:26:bf 
Found Device :-)
Found a printer!, connecting...
 - Created client, connecting to fc:58:fa:33:d7:4c
Got data transfer characteristic!
Testing plain text printing
Disconnecting
Done!
```

However it still be silent.

bitbank2, 2021-08-19:
The printer language must be different; sorry I can't do anything else without access to that printer.

closed 2021-08-19 by bitbank2