bitbank2/Thermal_Printer #11: Unable to get the service for PeriPage Mini A6 Thermal Printer (Arduino Nano 33 BLE Sense)

antigones, 2021-09-03:
Branch: Experimental
Printer: Peripage
Board: Arduino Nano 33 BLE Sense

Tried to use the driver but couldn't get the service:

Found a printer!, connecting...
connection attempt...
Connected!
Service 0x⸮F discovered
Trying to get service
Didn't get the service
connection failed

bitbank2, 2021-09-04:
I just tested this with the Nano 33 BLE and got this result:
Scanning for BLE printer
Scanning without a specific name
Found 4f:64:28:47:40:89 '' 
Found 20:21:01:12:c0:3c 'PeriPage+C03C_BLE' fee7
Found a match for PeriPage+
Printer type = 3
Found a matching device - PeriPage+C03C_BLE
Found a printer!, connecting...
connection attempt...
Connected!
Service 0xff00 discovered
Trying to get service
Got the service
Got the characteristic
Connected!
Printer pixel width = 576
Reported printer name = PeriPage+C03C_BLE
Preparing image buffer...
Disconnecting
Done!

Make sure you're using the latest code. I'm using a PeriPage+ (the 304dpi version also called the A6). Is yours the 203dpi version? Maybe the service/char changed.

antigones, 2021-09-04:
Just checked and yes, it's the 203dpi version.
Probably that's the issue.

bitbank2, 2021-09-04:
Use an app such as nRF Connect to explore the Service/Characteristics available and let me know. I'll try changing the numbers to match. Can you please add screenshots from nRF Connect if you're unsure which ones are the right ones?

antigones, 2021-09-04:
I basically found that hard-coding the service and characteristic UUID works and connection + discovery + service/characteristic get is successful. Yet the printer does not print anything. 🤷‍♀️

if (peripheral.discoverService("ff00")) {

pRemoteCharacteristicData = prtService.characteristic("ff02");
if (pRemoteCharacteristicData) {
...
}
}

I tried with both my "writable" characteristics. Maybe the protocol is not same here? 😢

bitbank2, 2021-09-04:
It's very possible that the protocol is different. The PeriPage does things a little differently than other BLE thermal printers.

closed 2021-09-04 by bitbank2

antigones, 2021-09-05:
I found this python snippet for 203dpi communication anyway: https://github.com/linglingltd/peripage-a6-control/blob/main/ppa6ctl.py

bitbank2, 2021-09-05:
That's the same protocol and bytes I'm already using to communicate with the PeriPage+