bitbank2/Thermal_Printer #12: CAT printer not working

TheNitek, 2021-09-09:
I trying to get my ESP32 to print on my brand new cat printer. I know it's supposed to crash, but it doesn't. Instead the printer just doesnt print (I tried all 3 example projects).

Output for Thermal_Printer_Demo looks like this:

```
Found a compatible device - GB02
Found a printer!, connecting...
 - Created client, connecting to 2e:7f:32:xx:xx:xx
Came back from connect
 - Found our service
[E][BLERemoteCharacteristic.cpp:287] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[E][BLERemoteCharacteristic.cpp:287] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[E][BLERemoteCharacteristic.cpp:287] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
Got data transfer characteristic!
Connected!, printing graphics
Testing plain text printing
Disconnecting
Done!
```

Printer LED stops blinking for a while, but nothing else happens. I am using the current master of this repo.

Any hints where to look and/or how to debug this?

bitbank2, 2021-09-09:
It may be that it's in a "semi-working" state instead of crashing, but the same problem exists. I really don't know how to get around it. Do you have an Arduino Nano 33 BLE or Adafruit nRF52 board to try? Those will work with the cat. I wish I had a better answer for you, but there is definitely something wrong with the ESP32 BLE library code (not the hardware).

closed 2021-09-09 by bitbank2

TheNitek, 2021-09-09:
I was planning to try using NimBLE instead of the Arduino-ESP implementation. Do you know if this has a chance to succeed?
Actually I wanted to dig deeper into the crash, but now there seems to be nothing to dig into 😬

bitbank2, 2021-09-09:
I tried NimBLE and got the same results.