bitbank2/Thermal_Printer #16: Phomemo T02 support

DieterKoblenz, 2021-11-15:
Is there a way to get the Phomemo T02 supported?

NB: I'll be analyzing the T02 to see if it can be included, unless someone already got it working. From the outside it looks similar to the cat devices, but it's not quite the same. Generic android BLE thermal apps can use the device, so I've got hope.

bitbank2, 2021-11-15:
If it uses the "standard" thermal printer command set, then it should be easy to add. Do you have it in your possession? If so, can you get the BLE advertised name and service/characteristic UUIDs?

DieterKoblenz, 2021-11-16:
UTF8 name: `T02`

Service UUIDs:

`0000FF01-0000-1000-8000-00805F9B34FB` notify
`0000FF02-0000-1000-8000-00805F9B34FB` write
`0000FF03-0000-1000-8000-00805F9B34FB` notify

bitbank2, 2021-11-16:
I pushed an untested change to add support for your printer. Please do a git clone of this repo and try it.

DieterKoblenz, 2021-11-17:
`const uint8_t ucBLETypes[] = {PRINTER_MTP2, PRINTER_MTP3, PRINTER_MTP3, PRINTER_CAT, PRINTER_CAT, PRINTER_CAT, PRINTER_CAT, PRINTER_CAT, PRINTER_PERIPAGEPLUS, PRINTER_PERIPAGE, PRINTER_PHOMEMO};`

DieterKoblenz, 2021-11-17:
It works, only the example prints too short, only half the arduino logo is printed.

bitbank2, 2021-11-20:
Can you send a photo? Is it that it doesn't print the last lines or that it doesn't feed the last lines?

DieterKoblenz, 2021-11-21:
[image]
This is after running the program.

[image]
This is the full printed picture.

bitbank2, 2021-11-26:
The fact that the top half printed successfully makes me think that you defined the graphics "back buffer" to be half the height it needs to be. Otherwise the pixels/logo would be garbled if something else went wrong. Can you provide the whole example code?

cqx931, 2021-12-03:
@DieterKoblenz What's the generic android BLE thermal apps you used to test the printer? I have a thermal printer from dodocool using the app [starpany](https://play.google.com/store/apps/details?id=com.starpany.app&hl=en&gl=US). I would  like to test whether I can use this library with it.

cqx931, 2021-12-07:
I also tried this and despite the bmp part, the tpPrint plain texts are also not printed.

raychmond, 2022-01-05:
I also tried on my Phomemo M02S, the 304dpi version. I added my model to the szBLENames and ucBLETypes. It can connect successfully. My service UUID is different from T02, which is "fee7". I don't know how to get the characteristic UUID. But it seems that it doesn't matter what UUID I use. 

The result is similar to others have seen, half arduino after several reboots.  I couldn't print plain texts at all.

Do you have plan to support Phomemo models?

Thank you!

bitbank2, 2022-01-05:
I wasn't planning on buying any more BLE thermal printers, so the answer is probable no :(

taylorjdlee, 2023-06-08:
@raychmond @DieterKoblenz our repo https://github.com/vivier/phomemo-tools is aiming to support as Phomemo printers as possible if you can collect the packets for the T02 and M02s that'll be great