bitbank2/Thermal_Printer #17: Support dodocool printers

cqx931, 2021-12-05:
I am trying to get my dodocool printer to work with the library. Could you please add it to the library?
`Name: starpany200, Address: dc:0d:30:97:8c:83, manufacturer data: 0000dc0d30978c83, serviceUUID: e7810a71-73ae-499d-8c15-faa9aef0c3f2 `

cqx931, 2021-12-05:
I tried to add the printer myself here:
https://github.com/cqx931/Thermal_Printer/commit/8c9232ae5b976cd83a8cad720e33f0d09b98b26e
So far I can get the connection established but it doesn't print anything. Is there anything I missed or got it wrong?

bitbank2, 2021-12-05:
If it uses the standard "ESC" command set, then you should not create a new type of printer. Add the name, but reuse the MTP2 printer type. Test to see that it works like a "normal" printer with the NRF Connect app on your phone. Write a short string of hex and terminate it with a CR; that should produce print output if you're using the correct service/characteristic.

cqx931, 2021-12-06:
Thank you for the reply! 

Could you give me an example hex string that I can try? I'm not very familiar with hex and ESC/POS command set and I'm afraid that I might tried the wrong commands. So far I tried to send some random utf-8 string or the line feed command "0A" in hex directly using an BLE Scanner app but I didn't see any response. I also tried it with other services I see under the device but no luck so far.

[image]

bitbank2, 2021-12-06:
Try using the 0x18f0 service and see if there's a write-only characteristic within it. Send the hex string: 0x41 0x42 0x43 0x0D

This should print "ABC" and advance the page 1 line

cqx931, 2021-12-06:
I tried it and still nothing happened. Maybe the model does not support the ESC/POS command set. I will try with another model tomorrow.

Thank you for your help and the amazing library!

bitbank2, 2021-12-06:
Try the other service UUID too. The characteristic you want will have Write, but not Notify.

cqx931, 2021-12-07:
So today I tried it with Phomemo T02 and the library works with esp32. However, I still can't get it working with the app with the above command even with this model, which gives me hope that my other printer could potentially work with the library... 

Do you have an app suggestion one should use? I'm trying to understand what else I could potentially messed up here.

raychmond, 2022-01-05:
> So today I tried it with Phomemo T02 and the library works with esp32! However, I still can't get it working with the app with the above command even with this model, which gives me hope that my other printer could potentially work with the library...
> 
> Do you have an app suggestion one should use? I'm trying to understand what else I could potentially messed up here.
> 
> This is what I tried with T02:
> 
> [image]

@cqx931 
I have a similar printer (Phomemo M02S). I tried different things. It can connect, but doesn't print correctly. Do you mind if you share what you did on your T02? I'm using ESP32 as well. Thank you so much!