bitbank2/Thermal_Printer #36: Phomemo t02 problem

yapudjus, 2022-04-09:
when trying to connect, it doesn't want to and give me this error

```3:04:24.804 -> ets Jun  8 2016 00:22:57
13:04:24.804 -> 
13:04:24.804 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
13:04:24.804 -> configsip: 0, SPIWP:0xee
13:04:24.804 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
13:04:24.804 -> mode:DIO, clock div:1
13:04:24.804 -> load:0x3fff0018,len:4
13:04:24.804 -> load:0x3fff001c,len:1216
13:04:24.804 -> ho 0 tail 12 room 4
13:04:24.804 -> load:0x40078000,len:10944
13:04:24.804 -> load:0x40080400,len:6388
13:04:24.837 -> entry 0x400806b4
13:04:25.135 -> Preparing image buffer...
13:04:25.135 -> Scanning for BLE printer
13:04:30.848 -> Found a printer!, connecting...
13:04:32.342 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
13:04:32.375 -> Core 1 register dump:
13:04:32.375 -> PC      : 0x4017be03  PS      : 0x00060730  A0      : 0x800d20fd  A1      : 0x3ffcfa70  
13:04:32.375 -> A2      : 0x00000000  A3      : 0xc12dbf68  A4      : 0x3ffccb4c  A5      : 0xc12dbf68  
13:04:32.375 -> A6      : 0x3ffe74cc  A7      : 0x00010000  A8      : 0x800d62e8  A9      : 0x3ffcfa10  
13:04:32.375 -> A10     : 0x3ffcfa5c  A11     : 0x3ffe74cc  A12     : 0x00000024  A13     : 0xc12dbf68  
13:04:32.409 -> A14     : 0x00000000  A15     : 0x0000000a  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
13:04:32.409 -> EXCVADDR: 0x00000013  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
13:04:32.409 -> 
13:04:32.409 -> ELF file SHA256: 0000000000000000
13:04:32.409 -> 
13:04:32.409 -> Backtrace: 0x4017be03:0x3ffcfa70 0x400d20fa:0x3ffcfa90 0x400d237d:0x3ffcfad0 0x400d14c7:0x3ffcfaf0 0x400d832a:0x3ffcfb20 0x4008ff66:0x3ffcfb40
13:04:32.442 -> 
13:04:32.442 -> Rebooting...
13:04:32.442 -> ets Jun  8 2016 00:22:57
13:04:32.442 -> 
13:04:32.442 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
13:04:32.442 -> configsip: 0, SPIWP:0xee
13:04:32.442 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
13:04:32.442 -> mode:DIO, clock div:1
13:04:32.442 -> load:0x3fff0018,len:4
13:04:32.442 -> load:0x3fff001c,len:1216
13:04:32.442 -> ho 0 tail 12 room 4
13:04:32.442 -> load:0x40078000,len:10944
13:04:32.442 -> load:0x40080400,len:6388
13:04:32.475 -> entry 0x400806b4
13:04:32.774 -> Preparing image buffer...
13:04:32.774 -> Scanning for BLE printer
```

yapudjus, 2022-04-09:
after a bit of searching, i found that the problem come from the line `            if(pRemoteCharacteristicNotify->canNotify())`
still dunno how to fix it, tried replacing it with `if(true)` to test the next line `pRemoteCharacteristicNotify->registerForNotify(ESP_notify_callback);` and i still got an error, but this time it was 
```Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d5404  PS      : 0x00060730  A0      : 0x800d5461  A1      : 0x3ffcf980  
A2      : 0x000000fc  A3      : 0x00000024  A4      : 0x3ffccb4c  A5      : 0x49bd7489  
A6      : 0x00000000  A7      : 0x0000000a  A8      : 0x400d15f0  A9      : 0x3ffcf960  
A10     : 0x00000000  A11     : 0x400d15f0  A12     : 0x00000002  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000100  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x400d5404:0x3ffcf980 0x400d545e:0x3ffcf9c0 0x400d2181:0x3ffcfa10 0x400d23f1:0x3ffcfa50 0x400d14df:0x3ffcfa70 0x400d844e:0x3ffcfaa0 0x4008ff66:0x3ffcfac0
```
still no clues at how to fix it

bitbank2, 2022-04-10:
I looked into this before and found the problem was in the BLEDiscoverServices() code. Either way, it's in the ESP32 support and I can't see an easy way to work around it.

closed 2022-04-10 by bitbank2

yapudjus, 2022-07-07:
> Sorry, but I had this on two ESP32 models. It used to work on an ESP 32...
> 
> @yapudjus did you ever find a way to get it working? 
> 
> @bitbank2 is there a  supported model on which it works?

never made it work, I'll buy an esp8266 and try something else