bitbank2/Thermal_Printer #37: CAT Printer YHK-B9AA

abhigkar, 2022-07-22:
I have another 'CAT' printer but it advertised as YHK-B9AA. This awesome library is not working with my cat printer. I tinkered the code and tried several other hacks using nrfConnect app. Here are my observations.

1- Change the [this line](https://github.com/bitbank2/Thermal_Printer/blob/master/src/Thermal_Printer.cpp#L80) with `{(char *)"YHK-B9AA", PRINTER_CAT},`. Printer found but with "Data service not found" error.  

2- It looks like the SERVICE_UUID and CHAR_UUID_DATA are not matching with the printer . I have used nrfConnect to get the UUIDS for Service and Characteristic. Surprisingly nrfConnect shows the same UUIDS that are defined for other types of printer.
                                     static BLEUUID SERVICE_UUID1("49535343-FE7D-4AE5-8FA9-9FAFD205E455");
                                     static BLEUUID CHAR_UUID_DATA1 ("49535343-8841-43f4-a8d4-ecbe34729bb3");

Esp32 crashing and rebooting 

`01:45:40.135 -> Found a compatible device - YHK-B9AA
01:45:40.135 -> Found a printer!, connecting...
01:45:40.135 ->  - Created client, connecting to 71:2b:0b:07:c4:ad
01:45:40.510 -> Came back from connect
01:45:41.495 ->  - Found our service
01:45:41.495 -> 
01:45:41.495 -> Stack smashing protect failure!
01:45:41.495 -> 
01:45:41.495 -> abort() was called at PC 0x40161bdf on core 1
01:45:41.495 -> 
01:45:41.495 -> ELF file SHA256: 0000000000000000
01:45:41.495 -> 
01:45:41.495 -> Backtrace: 0x4008ec74:0x3ffd0900 0x4008eef1:0x3ffd0920 0x40161bdf:0x3ffd0940 0x400d62c6:0x3ffd0960 0x400d62ee:0x3ffd0a20 0x400d2182:0x3ffd0a80 0x400d24e1:0x3ffd0ac0 0x400d155f:0x3ffd0ae0 0x400d8502:0x3ffd0b10 0x4008ff66:0x3ffd0b30
01:45:41.495 -> 
01:45:41.495 -> Rebooting...
`
AND Sometimes 
`lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)` I am sure it is not random but...

3- Follow [this issue](https://github.com/bitbank2/Thermal_Printer/issues/17). Issued some commands 0x41 0x42 0x43 0x0D to one of the WRITE only characteristic. Nothing printed but printer responded and feed the paper for single line. Simply 0x0D did the same thing.
Searched for ESC_POS commands and found [Line feed](https://escpos.readthedocs.io/en/latest/cursor_position.html#x0a) and [Carriage return](https://escpos.readthedocs.io/en/latest/cursor_position.html#x0d) is working with same enhancing the paper.

Here I am bit confused about what protocol is there in my 'cat'. Is there any way to further debug? 
For ESP32 crashing, I have Particle Xenon nrf5280 and Sparkfun nrf52832 breakout board. None of them are from Adafruit So cannot use then and ESP32 is the only option I have.

[image]

[image]

bitbank2, 2022-07-22:
I am stuck with that printer as well. I tried several ideas, but the protocol is purposely strange to prevent using it without their app. I tried decompiling their Android app to see if I could gain some insights, but wasn't able to get anything useful from it. I've now got 2 of these printers (MX06 cat/rabbit) and the YHK-xxxx one that I can't communicate with.

abhigkar, 2022-07-22:
Oh!! that is sad. By the way BLE sniffing may help. But I don't have nrf53 dongle fir that sake,

abhigkar, 2022-07-25:
I somehow figured it out that the YHK devices are also publishing classic Bluetooth MAC addresses. When I did an RFCOMM serial connection, the printer responded well. After analyzing btsnoop_hci.log from an android phone, it verifies that the WalkPrint app is also not using LE protocol to communicate with the printer. See attached ZIP file which can be opened in Wireshark. Just put a filter "Bluetooth.addr==a1:d1:e2:00:b9:aa" to narrow down the search.

I also replay the data payload on rfcomm serial and got the same result as I was getting from the android app.
The app sends everything as bitmap data to the printer and that is the saddest part as it is difficult to reconstruct the logic, though I have found the decompiled functions from the APK, but I am not skilled to decode in Python or another language.

[20220724200243.zip](https://github.com/bitbank2/Thermal_Printer/files/9182955/20220724200243.zip)

abhigkar, 2022-07-25:
> I am stuck with that printer as well. I tried several ideas, but the protocol is purposely strange to prevent using it without their app. I tried decompiling their Android app to see if I could gain [...]

I can help you to navigate the APK decompiled function if you say

mattheys, 2022-09-02:
Is there any progress with this? I've just got a MX06 I'd like to get working too.  I'll take a look at the zip,  did anyone get any further with the bitmap data?

toastmanAu, 2023-07-08:
> Is there any progress with this? I've just got a MX06 I'd like to get working too.  I'll take a look at the zip,  did anyone get any further with the bitmap data?
> 
> 
> 
> Did you print something that looked like this, if not can you share what you printed with that dump please?  I think there is possibly some header info in the packets, as I joined 3 of your packets out of the wireshark dump which is causing the top and bottom of the image to not be aligned correctly.
> 
> 
> 
> [image]
> 
> 
> 
> 
> 
> [Edit]
> 
> My Cat MX06 works with this library https://github.com/TheNitek/CatGFX/  not sure what is different.

Further to this have had some success with the catGFX library. Had to turn WiFi off for it to communicate. Also had to connect to my MX06 passing the MAC address in as it wouldn't auto detect it.

closed 2023-07-12 by abhigkar