bitbank2/Thermal_Printer #42: ESP32 connecting, but not quite connecting

davide1a, 2023-02-06:
Hi, I'm trying to use this library on an ESP32 with a Goojprt MTP-3, but can't seem to get it to connect properly. If I just use the `if (tpScan())` it doesn't find it at all. If I use `if (tpScan((char *) "MTP-3", 5))`, then it finds the printer and tries to connect - Serial monitor prints the "Found a printer!, connecting..." but then nothing happens.
It seems like it is connecting because after it finds the printer, the blue LED lights on the printer showing a bluetooth connection, and I then can't connect another device (my phone) to it. But presumably isn't quite connecting properly to actually print anything.

Any ideas?

bitbank2, 2023-02-06:
If the blue LED lights up and stays on, the connection should be solid. The printer's communication protocol may not be compatible and that would prevent it from working beyond connecting. Try connecting to it from a simple BLE app like Nordic's nRF Connect and just sending it ASCII text followed by a carriage return. It should print simple text directly. If that works, then I'm not sure what the problem is without seeing the rest of your code.

davide1a, 2023-02-06:
Thanks for the quick reply.
I got the nrf connect app (on my phone, is that what you meant?) connected
to the printer, but to be honest I really don't understand any of it :(

I've attached some screenshots of what it shows, but I don't know how I
should send it anything to print.

If it's any help, I have another Bluetooth printer app that works with it,
and the code I'm running is just the demo code from the library.

bitbank2, 2023-02-06:
hmm - if you're running the code successfully on another printer, then it sounds like the protocol of the failing one is different. If you're not familiar with how the code works, then it will be challenging to diagnose the problem remotely. Printer vendors either use the standard printing protocol (like the MTP-2) or they go it on their own. Can you send a purchase link to the one you're having trouble with? Maybe I'll buy one to figure out the issue.

davide1a, 2023-02-06:
Sorry, I didn't mean the code was working on another printer. I meant I have another app on my phone (just called Bluetooth Print) that works with this printer.
The printer is one of these: [https://www.tomtop.com/pt/p-os1228uk.html?currency=EUR&Warehouse=CN&aid=gttpmaxptth&gclid=Cj0KCQiA54KfBhCKARIsAJzSrdoLMBOW7jGu-iI_cN29RniFMp02ffVEvifo-S74sCrAGvIVnX3cy9gaAr_nEALw_wcB](url)

[image]
[image]
[image]

bitbank2, 2023-02-06:
I have and support that printer; I forgot because I have so many. Now I'm really curious as to why you're unable to get it working.

bitbank2, 2023-02-06:
Looks ok. I need to retest. Each time Espressif releases an updated support package for Arduino they break something. What version of ESP32 support are you using? I'll retest with your version to see if I can reproduce the problem.

davide1a, 2023-02-06:
It's on 2.0.6

I can try an older version maybe.

<https://www.e1a.pt>

davide1a, 2023-02-06:
I've tried the previous few versions, and it seems to be the same.

bitbank2, 2023-02-06:
ok, thanks for the info. I'll test on my equipment.

davide1a, 2023-02-06:
Thanks a lot. Hope you manage to find something!

If it helps, I've been playing a bit more, and with the demo sketch it just stops at "found printer, connecting" so I tried adding the following into the main loop, just to see if it was freezing completely, or if it was still running and maybe would try to connect again:

`void loop() {
  // put your main code here, to run repeatedly:
delay(5000);
  if (tpIsConnected()) {
    Serial.println("Still Connected");
    tpSetFont(1, 0, 0, 0, 0);
    tpPrint((char *)"9x17 plain text\r");
  }
  else if (!tpIsConnected()) {
    Serial.println(tpIsConnected());
    tpConnect();
  }
}`

With this, it does carry on trying, and after a few attempts, it does say "connect failed":
18:29:38.516 -> Scanning for BLE printer
18:29:44.184 -> Found a printer!, connecting...
18:29:49.742 -> 0
18:29:54.739 -> 0
18:29:59.733 -> 0
18:29:59.733 -> Connect failed
18:30:04.722 -> 0
18:30:04.722 -> Connect failed
18:30:09.739 -> 0
18:30:09.739 -> Connect failed
18:30:14.751 -> 0
18:30:14.751 -> Connect failed

So, it seems to be that the printer thinks they are connected, but that information isn't getting sent back to the esp for some reason.

bitbank2, 2023-02-08:
I just tried it with my MTP-3 and an Arduino Nano 33 BLE. I added the name in the tpScan() method and it immediately found the printer and successfully printed.

davide1a, 2023-02-08:
This is with an esp32 though. I don't have any Bluetooth adapters for the
arduinos.

bitbank2, 2023-02-08:
I just tested it with the ESP32 (generic LOLIN32 board), and it worked perfectly.

davide1a, 2023-02-08:
Maybe I'll try a different printer then. Still don't understand where the
extra lines were coming from in the serial monitor though.

davide1a, 2023-02-09:
Looks like I might struggle to get another one locally. In the meantime, if you get bored and if it helps, I loaded the demo code again with the core debug mode set to debug and verbose, so I don't know if those might show any reason for it not working?

[esp32_thermal_printer_debug.txt](https://github.com/bitbank2/Thermal_Printer/files/10698007/esp32_thermal_printer_debug.txt)
[esp32_thermal_printer_verbose.txt](https://github.com/bitbank2/Thermal_Printer/files/10698008/esp32_thermal_printer_verbose.txt)

aeschi, 2023-07-18:
Hi there! I have a similar problem. I am using an [ESP32 NodeMCU Development Board](https://www.berrybase.de/en/esp32-nodemcu-development-board) and I am able to find the printer (G00JPRT PT-210). The light of the printer also turns from red to blue/purple so seems to be connected, but nothing is printing. I am using the thermal printer demo file to test and I also use the tpScan() function with a specified name like so: tpScan("PT200",10)

This is the serial monitor output I get:

14:34:03.826 -> Scanning for BLE printer
14:34:14.532 -> Found a printer!, connecting...
14:34:21.648 -> Came back from connect
14:34:26.629 -> 0
14:34:26.629 -> Came back from connect
14:34:31.658 -> 0
14:34:31.658 -> Came back from connect
14:34:36.643 -> 0
14:34:36.643 -> Came back from connect
14:34:36.643 -> Connect failed
14:34:41.623 -> 0
14:34:41.623 -> Came back from connect
14:34:41.623 -> Connect failed

and then it continues like this.

I used your Print2BLE drag and drop app to print and that worked fine.

Any suggestions on what to try next? Thank you!

bitbank2, 2023-07-18:
Check the version of the ESP32 Arduino or IDF library you're using. There were changes to the BLE support that broke some functionality. Try regressing to an older version.

davide1a, 2023-07-18:
I ended up giving up with this library and used the BLEDevice library
instead.

aeschi, 2023-07-19:
Thanks for the advice! Version 1.1.2 also worked perfectly for me.

oixtron, 2025-05-01:
I had the same problem when trying to connect to the printer, my printer also uses the naming style like "PT210_ABCD". The issue is, that the printer is not correctly identified as `PRINTER_MTP2` printer type when passing the name with `tpScan()`. I modified the source file by adding the proper printer name with the correct type to the `szPrinterIDs[]` array e.g. `{(char *)"PT210_ABCD", PRINTER_MTP2},` and also removing `szName[9] = 0;` in `static uint8_t tpFindPrinterName(char *szName)` which resolved all problems with the latest version of this library. After that you can/should use tpScan() and tpConnect() whitout passing printer names.