bitbank2/Thermal_Printer #44: peripage a2 very light text and images

Joshinken, 2023-02-25:
im using the peripage a2 printer, and it connects just fine (thought the mcu crashes when trying to load the printer width), all custom font text and all images are really light? text prints just fine as complete black, though the different fonts and bold and such dont work, but the moment you print an image, its barely visible
[image]

Joshinken, 2023-02-25:
honestly for what im doing i just need the "huge font" to work, thats all i need. i can look past every single other issue that comes from this specific printer as long as i can just get a darker image here

bitbank2, 2023-02-25:
Sorry, I'm not sure. Some printers have an "energy" setting which really sets how long the paper sits on the heating element. I don't know the full command set for the PeriPage, so I don't know how to set it (if it exists).

closed 2023-02-25 by bitbank2