bitbank2/Thermal_Printer #47: Crash on M5stack-atom

GianniAntini, 2023-05-09:
I have a MTP-II thermal printer, that is correctly found. When the istruction:

`       pRemoteService = pClient->getService(SERVICE_UUID0);
`

is executed i obtain a guru meditation error, with kernel panic:

```
Found a compatible device - MPT-II
Found a printer!, connecting...
 - Created client, connecting to dc:0d:51:bb:ad:34
Came back from connect
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (BTU_TASK) 
Core  0 register dump:
PC      : 0x400955d7  PS      : 0x00060f36  A0      : 0x80094ecb  A1      : 0x3ffd2150  
A2      : 0x3ffbf648  A3      : 0xb33fffff  A4      : 0x0000abab  A5      : 0x0000cdcd  
```

Any idea?

GianniAntini, 2023-05-10:
A little update:
I'm using platformio as IDE. Using:

`lib_deps = bitbank2/Thermal Printer Library@^2.1.0`

the printer is NOT recognized at all (in source code MPT-II in not present). With library code con github, a have the crash reported above...

bitbank2, 2023-05-11:
Sorry, I don't use platformio. I would appreciate if you can find the issues, but I cannot.

closed 2023-05-11 by bitbank2