bitbank2/Thermal_Printer #48: Niimbot support

kris0725pl, 2023-06-03:
D11 and B21 please ;)

bitbank2, 2023-06-04:
I have the Niimbot label printer and a BLE capture tool, but the tool only has software support for Windows (which I don't use). I ordered a cheap NUC and when it arrives, I'll try to reverse engineer the NIIMBOT protocol.

kris0725pl, 2023-06-04:
[image]
I dissssemble d11 model

There is two ic.

LP28400A and ET4435

On pcb we have jtag?
Tx rx 3v3 gnd res pwk 3v3 and small 4pin tx1 rx2 G V

Battery...two cells 7.4-8.4v

Control by LP28400A
[image]

bitbank2, 2023-06-05:
Are you going to probe the TX/RX for data traffic?

kris0725pl, 2023-06-05:
I try today...

123thecong, 2023-07-23:
Hello 
Is there a way to print on sharp 3rd labels, I have a D110

MultiMote, 2025-01-27:
Some dumps can be found here: https://github.com/AndBondStyle/niimprint/issues/31
But studying dumps is not enough, there are some pitfalls when you trying to support more than one model

I wrote some info here: https://multimote.github.io/niimbluelib/documents/NIIMBOT_printers_protocol.html