bitbank2/Thermal_Printer #50: PT-210 advertising

MYCAMEL222, 2023-06-15:
I just bought a PT-210 from Amazon. 

It does not advertise as a "PT-210". It advertises as PT-210_D938. The _D938 is the last four digits of the MAC address of this printer. I assume they did this so if multiple printers were in the same area, you would know which is which.

Is there a way to add to the library to look for a PT-210 and auto connect to it by adding the _last for digits of the mac?

bitbank2, 2023-06-16:
I can change the logic to look for the substring "PT-210" and ignore any characters which come after. Can you try modifying the search string to include your printer's actual name and see if the rest of the logic works correctly?

MYCAMEL222, 2023-06-16:
Thanks for the reply...!

I can get it to connect using if (tpScan((char *)"PT-210_D938", 10)) search. It connects and then I get the serial monitor message of "Came back from connect". Then nothing happens. I've been able to get the PT-210 to print via a BLE serial terminal from my phone. I've attached two screenshots showing the UUID settings from my a BLE scanner app...

I was also wondering about the supported printer names in the .cpp file. It doesn't mention the PT-210 series?

`// Names and types of supported printers
const char *szBLENames[] = {(char *)"MTP-2", (char *)"MTP-3",(char *)"MTP-3F",(char *)"GT01",(char *)"GT02",(char *)"GB01",(char *)"GB02", (char *)"YHK-A133", (char *)"PeriPage+",(char *)"PeriPage_",NULL};
const uint8_t ucBLETypes[] = {PRINTER_MTP2, PRINTER_MTP3, PRINTER_MTP3, PRINTER_CAT, PRINTER_CAT, PRINTER_CAT, PRINTER_CAT, PRINTER_CAT, PRINTER_PERIPAGEPLUS, PRINTER_PERIPAGE};`

[image]
[image]

Thanks for your help..........

MYCAMEL222, 2023-06-18:
I tested each version of the library. At version 1.1.2 and using (tpScan((char *)"PT-210_D938", 10)) to find the printer, the demo ran and completed perfectly. Every version past 1.1.2 will connect, but nothing prints....

bitbank2, 2023-06-18:
good info, thanks. Will check.

MYCAMEL222, 2023-06-18:
More info from the debug output. This is where things hang up.

13:23:44.621 -> Found a printer!, connecting...
13:23:44.655 -> [ 15835][I][BLEDevice.cpp:622] addPeerDevice(): add conn_id: 0, GATT role: client
13:23:44.655 -> [ 15836][D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_REG_EVT
13:23:44.700 -> [ 15842][D][BLEClient.cpp:178] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_REG_EVT
13:23:44.765 -> [ 15928][D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_CONNECT_EVT
13:23:44.765 -> [ 15929][D][BLEClient.cpp:178] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_CONNECT_EVT
13:23:44.765 -> [ 15940][D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_OPEN_EVT
13:23:44.765 -> [ 15949][D][BLEClient.cpp:178] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_OPEN_EVT
13:23:44.808 -> Came back from connect
13:23:44.886 -> [ 16044][D][BLEClient.cpp:493] handleGAPEvent(): BLEClient ... handling GAP event!
13:23:46.036 -> [ 17214][D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
13:23:46.070 -> [ 17218][D][BLEClient.cpp:178] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown

MYCAMEL222, 2023-06-22:
In version 1.1.2, at the moment of connect, verbose debug spits out this .....

11:48:57.859 -> [ 12522][V][BLEClient.cpp:149] connect(): << connect(), rc=1

11:48:57.859 -> [ 12527][V][BLEClient.cpp:412] getService(): >> getService: uuid: 49535343-fe7d-4ae5-8fa9-9fafd205e455

Demo sketch will complete with 1.1.2

Version 2.0 and above spits out.......

09:52:26.130 -> [ 12037][V][BLEClient.cpp:149] connect(): << connect(), rc=1

09:52:26.164 -> [ 12066][V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT
09:52:26.164 -> [ 12068][V][BLEUtils.cpp:1253] dumpGapEvent(): [status: 0, bd_addr: 86:67:7a:b3:d9:38, min_int: 6, max_int: 12, latency: 0, conn_int: 12, timeout: 300]
09:52:26.196 -> [ 12078][D][BLEClient.cpp:493] handleGAPEvent(): BLEClient ... handling GAP event!
09:52:27.379 -> [ 13283][V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 46
09:52:27.379 -> [ 13283][D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
09:52:27.379 -> [ 13295][V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 46
09:52:27.425 -> [ 13295][V][BLEUtils.cpp:1283] dumpGattClientEvent(): GATT Event: Unknown
09:52:27.425 -> [ 13306][V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 46

It looks like 2.0 and above is having a hard time providing the ServiceUUID of the ESP32? 

I'm hoping there is a way around this.

bitbank2, 2023-06-23:
I'm sorry, but I have no idea.

closed 2023-06-23 by bitbank2