bitbank2/Thermal_Printer #51: OVERRIDING PASSWORD FOR SETTING UP CONNECTION WITH PRINTER

aishwarya2208, 2023-06-21:
Hi, I purchased a MTP-II printer as well as a PT-210 printer from amazon that require entering a password (either 0000 or 1234) inorder to be able to connect to it. I always get a Guru meditation error after the printers are identified and connection is established. Can I please get some help on this? 
[image]

aishwarya2208, 2023-06-21:
Thanks, I'll try this out. Could I also get a solution for overriding the password required by the printers during connection establishment?

bitbank2, 2023-06-21:
I don't know how to do that.