bitbank2/Thermal_Printer #53: Unable to connect to MTP-II

zejtin, 2023-08-02:
Hello!

I have an MTP-II printer which should be fine. Name on bluetooth is **MTP-II_BC6A**
I am trying to scan it with `tpScane((char *)"MTP-II_BC6A", 15);` to no avail, tpScan returns false but printer changes led color to blue like when it is connected when running tpScan.

Any info on this? 

Thanks in advance

MYCAMEL222, 2023-08-02:
use library version 1.1.2. It should work when you scan tpScan((char *)"MTP-II_BC6A", 15);

zejtin, 2023-08-03:
Tried it just now, it still doesn't find it, difference with 1.1.2 is printer doesn't show there is connection as with the latest one

MYCAMEL222, 2023-08-03:
Your code line in the above post reads "tpScane((char *)"MTP-II_BC6A", 15);".........It should be "tpScan....".

zejtin, 2023-08-03:
My bad on that, it is `tpScan` otherwise it wouldn't compile

MYCAMEL222, 2023-08-03:
I ended up writing my own code without the Thermal_Printer library. I used arduinoBLE library and scanned for UUID.  Everything works fine that way, but you'll have to hard code the esc/pos commands. 

Sorry I can't help more.

zejtin, 2023-08-04:
Do you maybe have some repo or example how you did it?

controllercustom, 2023-08-08:
To add a new printer name, truncate it to 9 chars and add it to szPrinterIDs[].

tpFindPrinterName() truncates the advertised printer name to 9 characters or less before looking for a match in szPrinterIDs.