bitbank2/Thermal_Printer #68: Does it work with MXW01?

eerimoq, 2025-02-08:
Hello!

I've implemented support for thermal printers in my project [Moblin](https://github.com/eerimoq/moblin). It works well, but a new printer model MXW01 does not work. I don't have an Arduino available, so figured I'll just ask here if someone knows how to communicate with an MXW01 printer?

eerimoq, 2025-02-12:
It works now! MXQ01 uses a totally different protocol compared to MX10. It's not pretty, but here is the code in case you want to implement it in this project as well:

https://github.com/eerimoq/moblin/blob/main/Moblin/Integrations/CatPrinter/CatPrinter.swift
https://github.com/eerimoq/moblin/blob/main/Moblin/Integrations/CatPrinter/CatPrinterCommandsMxw01.swift

eerimoq, 2025-02-13:
It seems to overheat a lot. Not sure how to make it more reliable :(

Paffo, 2025-02-20:
I also have this MXW01, the main characteristic is that the commands start with 0x22 0x21 instead of all other printers starting with 0x51 0x78.
Unluckily enough, the app that came with mine has the other printers as compatible too, so i tried decompiling it to extract the commands for ours, but no luck:
while the v5g printers have their commands hardcoded, this one fetches them from an API or something, i could find a lot of dynamic calls to getData() but no hard coded command like the DataBeans or similar.
I also used wireshark to intercept the BLE packets and i can confirm that they start with 0x22 0x21, but other than identifying the one for print quality (in percentage 0 to 100) i didn't manage to pin down the basics for printing.

In the next days i will try to convert that Swift one, let's see.
If anyone has any input please let me know, this is my first attempt at something like this

tytytytyy, 2025-05-04:
Do u have answers now? I have a connection but it doesnt print at all :(