bitbank2/Thermal_Printer #71: Firmware flashed on ESP32 board  but not working on real printer device

Prathamesh2000132, 2025-04-29:
Hello Team,

I cloned the project repository and successfully ran the firmware project by connecting it to an ESP32 board. I generated the .bin file and created a .exe file with a Windows installer to flash the firmware.
The firmware successfully flashes and runs on the ESP32 development board using the generated .bin file and my custom installer.
When I connect the real thermal printer device and try to flash the firmware, the flashing does not succeed.

The same .bin file flashes correctly to the ESP32 board but does not work or flash on the actual printer hardware, even though the printer's name is added correctly in the firmware source.

I will attach the error screenshots below.

[image]

bitbank2, 2025-04-29:
Sorry, but it sounds like you're trying to do something that's unrelated to this library. This code generates text and graphics which can be transmitted over BLE to some thermal printers. I'm really not sure what you're doing, but it doesn't relate to this library:
ESP32 firmware issues - not related
Thermal printer serial port issues - not related
Windows EXE files - not related

closed 2025-04-29 by bitbank2

Prathamesh2000132, 2025-04-30:
I have a C19 Mini Pocket Printer from the Seznik brand, but it's not connecting via BLE (Bluetooth Low Energy) except with specific apps like "Luck Jingle" and "SharpFox". It only connects with those apps, but I need it to connect with any app, especially apps related to the printer device.

Can you guide me on what needs to be done to make it connect immediately with any app? Could it be a firmware issue? If so, can you suggest any ESP/POS-compatible firmware that could work with this printer model?

I need some guidance on this.

bitbank2, 2025-04-30:
What firmware? This library does not send firmware to the printer.